ros-infrastructure/rep #97: Alternative version of ros::param::param with return value

xperroni, 2015-03-30:
The template function `ros::param::param` sets a variable either to the value of a parameter, or if not available, a predefined default value. In cases where the variable is passed straight to a constructor or function call and then disregarded, more streamlined code could be written if the result value could be retrieved as a function return value, rather than through an output variable.

For example, the code below:

    std::string path;
    std::string format;
    double fps;

    param("~path", path, "video.mpg");
    param("~format", format, "MPEG");
    param("~fps", fps, 30.0);

    VideoRecorder recorder(path, format, fps);

Could be more succinctly written as:

    VideoRecorder recorder(
        param<std::string>("~path", "video.mpg"),
        param<std::string>("~format", "MPEG"),
        param<double>("~fps", 30.0)
    );

If an alternative definition of `ros::param::param` such as this was available:

    template<typename T>
    T param(const std::string& param_name, const T& default_val)
    {
      T param_val;
      param(param_name, param_val, default_val);
      return value;
    }

tfoote, 2015-03-30:
Please open this for discussion at: https://github.com/ros/ros_comm/issues this repo is targeted at higher level concepts within ROS more than language specific APIs.

closed 2015-03-30 by tfoote

xperroni, 2015-03-30:
Done. Sorry for the blunder.

tfoote, 2015-03-30:
np. There's a lot of repos in the ros ecosystem it takes a while to learn the right places. Thanks for the quick follow up.