ros-infrastructure/rep #107: ROS Jade on Ubuntu 14.04 On PPC

jacinc2, 2015-10-07:
I encountered an issue in ROS in trying to install Jade on Ubuntu 14.04 running successfully without errors on my PowerPC G4 tower.  I posted the error I get trying to install ROS :

http://answers.ros.org/question/218605/ros-jade-on-ubuntu-1404-e-unable-to-locate-package-ros-jade-desktop-full/ 

But my real question is... according to documentation there is no "REAL" architecture for ROS so why won't it work in "Ubuntu 14.04".. or is there an underlying issue I am unaware of.

Can it be done?.. can it run under PPC with 14.04.. or is there a modification that I should for?

Let me know if this is a small or large job to do.. can I use the SOURCEs and reconstruct it to run under that environment?.. I figured that as a robot OS, it should be flexible with ANY device or hardware interface.

tfoote, 2015-10-07:
There's no 'underlying issue' with running ROS on PPC it's just not a platform that we build binaries for. 

If you compile from source most parts of ROS should work on a PPC architecture. There may be some packages which make architecture assumptions in the ecosystem but most packages compile for ARM and x86 thus I would expect them to compile for PPC as well. 

If you run into problems please follow up with a question on answers.ros.org

closed 2015-10-07 by tfoote

jacinc2, 2015-10-07:
Thank you so much.. I will compile for PPC.. much appreciated..