ros-infrastructure/rep #128: email address for orphaned packages

dirk-thomas, 2017-04-26:
What email address should be used for orphaned packages? [REP 140](http://www.ros.org/reps/rep-0140.html#id10) currently recommends `ros-release@code.ros.org` which seems to be outdated.

tfoote, 2017-04-27:
If it's truely unmaintained I think a blackhole email would be correct. And use the name portion to suggest posting to https://discourse.ros.org/c/release or an official Orphaned Package page on the wiki with guidelines for adoption etc.

dirk-thomas, 2017-04-27:
What name would you suggest? I am not sure I understand what / how you want to embed the information.

tfoote, 2017-04-27:
```
<maintainer email="orphaned@ros.org">Orphaned Package No Maintainer: Please visit http://wiki.ros.org/OrphanedPackage for more info</maintainer>
```

Where orphaned@ros.org would just be received, and discarded.

Then create an a page like http://wiki.ros.org/OrphanedPackage which talks about how to adopt a package etc.

tfoote, 2017-04-27:
After discussing it a bit it seems a black hole is less useful than an archived mailing list. So I've setup a google group we can use as an archive: https://groups.google.com/d/forum/ros-orphaned-packages It's setup to allow posting but not subscriptions so it will be browse only.

dirk-thomas, 2017-04-27:
> So I've setup a google group ...

@tfoote Great. Can you please add other users similar as for the buildfarm groups. Thanks.

dirk-thomas, 2017-04-27:
Btw the information for how to take over maintenance of another package already exists: http://wiki.ros.org/MaintenanceGuide#Claiming_Maintainership

dirk-thomas, 2017-04-27:
Please see #129 for a patch for the affected REPs.

dirk-thomas, 2017-04-27:
I reopened this until [the comment](https://github.com/ros-infrastructure/rep/issues/128#issuecomment-297806526) has been addressed.

tfoote, 2017-04-27:
Invitations sent.

dirk-thomas, 2017-04-27:
@tfoote Can you please make other owner too in order to access settings etc.

tfoote, 2017-04-27:
As I said in the invitation you need to ping me to be upgraded to an owner after you've accepted the invitation.  I've added several now and don't know who else wants to get these emails. If anyone else wants access please email me or any of the others on the list now.

closed 2017-04-27 by tfoote

dirk-thomas, 2017-04-27:
When creating the next mailing list we might want to add everyone on the ROS team using "Direct add members" and make everyone an owner. This avoids the need for sending and confirming invitations.

It might also be good to set the delivery options for everyone to "no emails". Then everybody can opt in. Now everyone needs to opt out after accepting the invitation (which you need to accept to have the necessary permissions of an owner in the future).

PR #129 (2017-04-27): update information about orphaned packages
Fixes #128.