ros-infrastructure/rep #132: Clarification on CATKIN_IGNORE location

jwhendy, 2017-07-07:
Hello,

I had a package failing that I wanted to just ignore for the time being and looked around for how to do it. The top hit I got was this [ROS Answers post](http://answers.ros.org/question/54181/how-to-exclude-one-package-from-the-catkin_make-build/) which points to [REP-128](https://github.com/ros-infrastructure/rep/blob/19167e7d09a45226af0d33ff90ed1353ec1b8738/rep-0128.rst). The top of that is as follows:

```
workspace_folder/         -- WORKSPACE
  src/                    -- SOURCE SPACE
    CMakeLists.txt        -- The 'toplevel' CMake file
    package_1/
      CMakeLists.txt
      package.xml
      ...
    package_n/
      CMakeLists.txt
      package.xml
      ...
  build/                  -- BUILD SPACE
    CATKIN_IGNORE         -- Keeps catkin from walking this directory
```

Based on that, I tried various ways to put a `build/CATKIN_IGNORE` folder somewhere with no success. After some frustration, I found [this other post](http://answers.ros.org/question/208180/how-to-use-catkin_ignore-file-correctly/) which showed just putting it in the top level of `./package_n` (to use your terminology above). That worked for me.

The package that's failing happens to be part of a meta-package and neither of these worked:
```
~/catkin_ws/src/meta_pkg/build/CATKIN_IGNORE
~/catkin_ws/src/meta_pkg/failing_pkg/build/CATKIN_IGNORE
```

These worked:
```
~/catkin_ws/src/meta_pkg/CATKIN_IGNORE
~/catkin_ws/src/meta_pkg/failing_pkg/CATKIN_IGNORE
```

Perhaps REP-0128 is superseded, but I only found the same mention of `build/CATKIN_IGNORE` in REP-0133 when [searching this repo](https://github.com/search?q=org%3Aros-infrastructure+catkin_ignore&type=Code).

Should this structure/suggestion be updated or clarified, or am I not understanding something correctly?

dirk-thomas, 2017-07-07:
The file `build/CATKIN_IGNORE` is being created by the build tool automatically to prevent finding packages within the build space.

You might also want to search answers.ros.org where we ask users to raise questions like this. The following one should answer how to use the marker file: http://answers.ros.org/question/208180/how-to-use-catkin_ignore-file-correctly/

I have created #133 to include an example in the REP 128.

jwhendy, 2017-07-07:
I'm pretty new and feeling dense, as I don't understand what creates these automatically by the build tool.

I did search ROS Answers as mentioned in the issue post. That's actually why I created this, as the first answer pointed me to REP 128, which only shows this file in a `build/` dir and not in the package top level dir. I then cited the same answer you suggested (which also helped me understand how to use it).

Basically this issue is about asking if, indeed, REP 128 is not clear about manual usage of `CATKIN_IGNORE` such that the official docs (this site) could illustrate it vs. having users scour ROS Answers where the quality/details vary greatly.

The PR looks great, and that is exactly what I hoped for. Thanks!

PR #133 (2017-07-07): Catkin ignore example
Fixes #132.
closed 2017-07-07 by jwhendy