ros-infrastructure/rep #137: REP 12: where should REPs be submitted?

gavanderhoorn, 2017-09-03:
As the final bullet under *How to Use This Template*, REP 12 has:
```
- Send your REP submission to the ROS developers at ros-users@code.ros.org.
```

The mailing lists were then migrated to `@lists.ros.org`, but [mailman there](http://lists.ros.org/mailman/listinfo/ros-users) now states:

> This is the historical location of the ROS users mailing list. Please use http://discourse.ros.org/ to continue discussions.
> 
> Posts there will be archived here as well, but direct posting to this list has been turned off.

Should ROS Discourse now be listed as the place where REPs should be posted for initial review / discussion?

jack-oquin, 2017-09-04:
ROS Discourse seems reasonable to me.

dirk-thomas, 2017-09-05:
> Should ROS Discourse now be listed as the place where REPs should be posted for initial review / discussion?

:+1: It would be great if you could make a PR with the proposed fix.

clalancette, 2022-02-16:
This was fixed, so closing out.

closed 2022-02-16 by clalancette