ros-infrastructure/rep #156: REP 149: Add an optional 'file' attribution to the license tag

otamachan, 2018-02-02:
Hi,
I would like to propose adding an optional ``file`` attribution to the ``license`` tag in REP 149.
As I pointed out in https://github.com/ros-infrastructure/bloom/issues/456, some OSS licenses require provision of license and copyright text when a user redistribute the software in a source or binary form. I think the file attribution will help a user or a tool to find right license or copyright text easily.

Thanks in advance.

tfoote, 2018-02-03:
Your proposed change looks reasonable. Could you submit a PR with that change to REP 149 with the logic called out for review?

dirk-thomas, 2019-11-13:
Closing since it seems to be addressed in #157.

closed 2019-11-13 by dirk-thomas