ros-infrastructure/rep #167: REP-0003 Melodic: Which version of OpenCV to use?

clalancette, 2018-03-27:
From @tfoote in https://github.com/ros-infrastructure/rep/pull/160#issuecomment-375420063:

I'd like to revisit the discussion for OpenCV. In particular we updated targets for everything to be 3.4 in #139 In particular comments at #139 (review) #139 (review)

We only pushed ahead to 3.3 based on this request: https://discourse.ros.org/t/opencv-3-3/2674 and that upgrade caused significant issues and debugging time. And we've had several request to fall back to 3.2 during the process. We thought that the new release would be fully backwards compatible but it was not and with the knowledge that I have now, I would have not chosen to make the upgrade if I could choose again.

With 3.2 available for our LTS platform (bionic) and 3.1 on the soon to be EOL'd platform (artful) the only major outlier is that stretch is older and only has 2.4.9.

I'd like to suggest that we fall back to using the system dependencies again instead of packaging OpenCV within ROS packages. On most of our platforms we're within a minor release.

To make things close to compatible I would propose that we backport buster's 3.2 to stretch, and then by July we'll only need to support 3.2 or higher. Which I believe is already available in Fedora and other platforms. And if there's something major blocking on Artful's 3.1 we could backport bionic's 3.2 or blacklist opencv3 on artful as it's very close to EOL.

There are significant advantages to using the upstream packages. In particular we are compatible with other system dependencies that rely on the system opencv installation. There are also standard replacement packages that enable hardware acceleration for systems with CUDA, Neon, etc maintained by the greater community. This would allow ROS users to leverage those builds.

We will also avoid significant build time on the ROS buildfarm, as well as maintaining a parallel custom package which has cost us a lot of time in this last cycle. In the past there's been issues with contrib not being available and the opencv version upstream being very out of date. Also as there's now the same major version of opencv on most platforms we need to be much more careful about declaring conflicts and making sure that our installation and the system one do not collide.

If people really do need to have the latest and greatest version of opencv, replacing the system version is relatively straight forward, it will just require that that any package depending on the custom version of opencv be compile from source. This is standard development processes, and with recent containerization technologies can even be relatively easily shared and reproduced.

I know that there are people that are strongly pushing for the bleeding edge latest things, but a large part of being in a distro is to be stable and solid for those not pushing the envelope. The new features will become available, but to provide the stability of a distro I think that we should not try to significantly push ahead of the upstream packaging.

@mikaelarguedas @vrabaud @clalancette

Follow-up from @clalancette:

I'm totally fine with this, as I agree the benefits of sticking to the system packages outweigh the benefits of having a newer package available. I'd even go further and suggest that we make sure to stick with whatever version we choose for the lifetime of the distribution, to make sure we don't have a repeat of what happened with OpenCV 3.3.

That all being said, I'm not one of the bleeding edge users that really want newer OpenCV features, so I'd like to hear from @vrabaud before we change this.

mikaelarguedas, 2018-03-27:
I'm generally in favor, there's on point I'd like to discuss below though.

Regarding shipping a custom version on Stretch, 3.2 is already in buster so it shouldn't be too hard to ship it.
There's a corner case that may be worth discussing. What happens if Debian Buster is not released when the next ROS LTS is released?
e.g. Ubuntu F ships with OpenCV 3.4 (or 4.0) and we ship 3.2 on Stretch based on this discussion.
What do we do for Stretch users that want to use the new ROS LTS distro?
- stick to 3.2 and force package maintainers to support both APIs?
- ship whatever is in buster at that time and thus forcing an upgrade on Melodic?
- find a way to package the 2 versions so that they're side by side installable without symbol collision? (which I expect to be significant effort)

> Follow-up from clalancette: I'd even go further and suggest that we make sure to stick with whatever version we choose for the lifetime of the distribution, to make sure we don't have a repeat of what happened with OpenCV 3.3.

Once we move to system dependencies this will be out of our control. Given Ubuntu policies for upgrade in stable distros, I would not expect to see a minor version bump in a released distro. But if this happens (it will be tested and maintained by Canonical) we should not prevent ourselves from following upstream as this would mean packaging and shipping an older version ourselves.

clalancette, 2018-03-27:
> I'm generally in favor, there's one point I'd like to discuss below though.
> 
> Regarding shipping a custom version on Stretch, 3.2 is already in buster so it shouldn't be too hard to ship it on Stretch.
> There's a corner case that may be worth discussing. What happens if Debian Buster is not released when the next ROS LTS is released?
> e.g. Ubuntu F ships with OpenCV 3.4 (or 4.0) and we ship 3.2 on Stretch based on this discussion.
> What do we do for Stretch users that want to use the new ROS LTS distro?
> 
>     stick to 3.2 and force package maintainers to support both APIs?
>     ship whatever is in buster at that time and thus forcing an upgrade on Melodic?
>     find a way to package the 2 versions so that they're side by side installable without symbol collision? (which I expect to be significant effort)

Would we really expect to support four different ROS1 versions on Debian Stretch (Lunar, Melodic, O-Turtle, and N-Turtle)?  Given Debian's slow release strategy, it is certainly not out of the range of possibility, but I'm not sure that's something we should do.  Regardless, we actually have the same problem no matter what we do here; we could end up choosing 3.4, and have Ubuntu F end up choosing 4.0, and then we'd been in mostly the same boat.  So while it is a problem, I think it is one we don't have any control over.

> Follow-up from clalancette: I'd even go further and suggest that we make sure to stick with whatever version we choose for the lifetime of the distribution, to make sure we don't have a repeat of [...]

Sure, I was assuming that Ubuntu would not bump the minor version during the lifetime of the distro, but I agree that once we choose the system packages we always stick with them.

vrabaud, 2018-04-12:
Hey, I'm the one to blame for the 3.3. mess. Your question is valid and there was no good solution:

Sticking to the distro version:
Pros: no maintenance, allows for CUDA supports (Tegra platforms have their own version)
Cons: not sure of what you get (some distros might not package contrib), old version

Having one ROS version:
Pros: reactivity to new releases (especially after GSOC or major ones (ROS had OpenCV3 before all Ubuntu distros for a year or two)
Cons: maintainer cannot test on all platforms

As I cannot find the time to maintain the ROS OpenCV package and as OpenCV did not get into GSOC this year, we might as well stick to 3.2 for this round. 4.0 will be out this summer with a lot of DNN stuff so the community might want it (just like it wanted 3.x while only 2.4 was available in Ubuntu).
Maintainers should be safe as long as they follow this rule: http://wiki.ros.org/opencv3#package.xml

tfoote, 2018-04-17:
3.3 is designed to be compatible with 3.2. Though we ran into challenges. We could revert the upgrade, but that would likely cause more disruption. I'd suggest that we clearly document that it's recommended to just use the 3.2 API and leave it as is.

clalancette, 2018-04-17:
> 3.3 is designed to be compatible with 3.2. Though we ran into challenges. We could revert the upgrade, but that would likely cause more disruption. I'd suggest that we clearly document that it's recommended to just use the 3.2 API and leave it as is.

Yeah, I definitely agree that we shouldn't revert Kinetic and Lunar, as that will cause further problems.  If we are OK with saying a "3.2" API, then we can just use document 3.2 across the board and go with that.  In that case, I'm going to go ahead and merge #168 , and then start building the OpenCV 3.2 packages for Artful and Stretch.

closed 2018-04-17 by clalancette

mikaelarguedas, 2018-04-17:
> 3.3 is designed to be compatible with 3.2. Though we ran into challenges. We could revert the upgrade, but that would likely cause more disruption. I'd suggest that we clearly document that it's recommended to just use the 3.2 API and leave it as is.

Agreed, breakage can be expected by moving either up or down. As long as we notify users of the downgrade and the rationale behind it I think it's fine

AeroWRX, 2019-08-05:
It's now August 2019, does it still hold true that we can only use OpenCv 3.2 on Melodic? or is there 3.4 support and also moving towards OpenCv4?

clalancette, 2019-08-05:
> It's now August 2019, does it still hold true that we can only use OpenCv 3.2 on Melodic? or is there 3.4 support and also moving towards OpenCv4?

Melodic still supports only OpenCV 3.2 (the version in Ubuntu Bionic), and unless something very unexpected happens, that should remain the case for the lifetime of Melodic.  When we release Noetic (May 2020), it will get the updated version of OpenCV that comes with the Ubuntu of the time (likely OpenCV 4 or later).