ros-infrastructure/rep #222: ros.org/reps very slow to load

gavanderhoorn, 2019-12-20:
(for me at least) it takes up to 10 seconds to load the HTML versions of REPs served by `www.ros.org/reps`. Doesn't seem to matter which REP.

Looking at the *Network* tab of the Developer Tools in Firefox seems to indicate that images are slow to load but the main `.html` also contributes.

I've experienced this on multiple machines with multiple different internet connections, but it could still be "local" of course.

tfoote, 2020-01-07:
#223 definitely might help. Can you provide a timing graph from the browser network for a page load? I'm seeing pretty quick load times myself, but I'm probably closer to the source and my browser is blocking all the css at the moment. 

[image]

A second reload mermaid's javascript is cached for me. But I pushed it later in the script in #224

gavanderhoorn, 2020-01-07:
I'm seeing similar nrs now:

[image]

mermaid is indeed the biggest contributor to loading time. I'm not familiar with it at all, is there a reason it's so massive? Could it perhaps be minimised?

gavanderhoorn, 2020-08-26:
Seems to have been fixed.

Loading times are acceptable now.

gavanderhoorn, 2020-08-26:
Thanks @tfoote :+1:

closed 2020-08-26 by gavanderhoorn