ros-infrastructure/rep #239: failing to build due to python2 sunsetting

tfoote, 2020-03-10:
This script is failing: https://github.com/ros-infrastructure/rep/blob/master/genrepindex.py

Called from the Makefile

http://build.ros.org/job/doc_independent-packages/287/console
<details>
22:46:54 # BEGIN SUBSECTION: rep: make html
22:46:54 + make html
22:46:54 mkdir -p _build/html
22:46:54 cd .. && make all
22:46:54 make[1]: Entering directory '/tmp/repositories/rep'
22:46:54 python genrepindex.py .
22:46:55 /usr/local/lib/python2.7/dist-packages/pkg_resources/py2_warn.py:22: UserWarning: Setuptools will stop working on Python 2
22:46:55 ************************************************************
22:46:55 You are running Setuptools on Python 2, which is no longer
22:46:55 supported and
22:46:55 >>> SETUPTOOLS WILL STOP WORKING <<<
22:46:55 in a subsequent release (no sooner than 2020-04-20).
22:46:55 Please ensure you are installing
22:46:55 Setuptools using pip 9.x or later or pin to `setuptools<45`
22:46:55 in your environment.
22:46:55 If you have done those things and are still encountering
22:46:55 this message, please comment in
22:46:55 https://github.com/pypa/setuptools/issues/1458
22:46:55 about the steps that led to this unsupported combination.
22:46:55 ************************************************************
22:46:55   sys.version_info < (3,) and warnings.warn(pre + "*" * 60 + msg + "*" * 60)
22:46:55 Traceback (most recent call last):
22:46:55   File "rep2html.py", line 87, in <module>
22:46:55     import docutils_readers_rep
22:46:55   File "/tmp/repositories/rep/docutils_readers_rep.py", line 13, in <module>
22:46:55     from docutils.parsers import rst
22:46:55   File "/usr/local/lib/python2.7/dist-packages/docutils/parsers/rst/__init__.py", line 75, in <module>
22:46:55     from docutils.parsers.rst import roles, states
22:46:55   File "/usr/local/lib/python2.7/dist-packages/docutils/parsers/rst/roles.py", line 78, in <module>
22:46:55     from docutils.utils.code_analyzer import Lexer, LexerError
22:46:55   File "/usr/local/lib/python2.7/dist-packages/docutils/utils/code_analyzer.py", line 19, in <module>
22:46:55     from pygments.formatters.html import _get_ttype_class
22:46:55   File "/usr/local/lib/python2.7/dist-packages/pygments/formatters/html.py", line 554
22:46:55     file=sys.stderr)
22:46:55         ^
22:46:55 SyntaxError: invalid syntax
22:46:55 Makefile:10: recipe for target 'rep-0010.html' failed
22:46:55 make[1]: *** [rep-0010.html] Error 1
22:46:55 make[1]: Leaving directory '/tmp/repositories/rep'
22:46:55 Makefile:6: recipe for target 'html' failed
22:46:55 make: *** [html] Error 2
22:46:55 Build step 'Execute shell' marked build as failure
22:46:55 $ ssh-agent -k
22:46:55 unset SSH_AUTH_SOCK;
22:46:55 unset SSH_AGENT_PID;
22:46:55 echo Agent pid 27593 killed;
22:46:55 [ssh-agent] Stopped.
22:46:55 Sending e-mails to: ros-buildfarm@googlegroups.com tfoote+buildfarm@osrfoundation.org
22:46:55 Finished: FAILURE
</details>

tfoote, 2020-03-10:
In the short term we could likely pin back to setuptools<45 as mentioned in the comment like we do for xmlschema: https://github.com/ros-infrastructure/ros_buildfarm_config/blob/7a3f87c752f7d3b64403cbafa5552e908b504e2a/doc-independent-build.yaml#L21

but we should also look at upgrading to python 3 overall.

chapulina, 2020-03-12:
I'm not sure if I followed all the tracks down to the correct place. Is this it? https://github.com/ros-infrastructure/ros_buildfarm/commit/be9e8832e120088caf0b12fef3bc20403ff0d4df

dirk-thomas, 2020-03-20:
http://build.ros.org/job/doc_independent-packages/298/

closed 2020-03-20 by dirk-thomas