ros-infrastructure/rep #250: HTTP 403 error

Tiryoh, 2020-04-21:
I cannot access to the contents.

```
$ curl -v http://www.ros.org/reps/rep-0000.html
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 2605:bc80:3010:104::8cd3:962:80...
* Connected to www.ros.org (2605:bc80:3010:104::8cd3:962) port 80 (#0)
> GET /reps/rep-0000.html HTTP/1.1
> Host: www.ros.org
> User-Agent: curl/7.69.1
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 403 Forbidden
< Date: Tue, 21 Apr 2020 04:00:55 GMT
< Server: Apache
< Content-Length: 281
< Content-Type: text/html; charset=iso-8859-1
<
{ [281 bytes data]
100   281  100   281    0     0   1072      0 --:--:-- --:--:-- --:--:--  1072<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>403 Forbidden</title>
</head><body>
<h1>Forbidden</h1>
<p>You don't have permission to access /reps/rep-0000.html
on this server.</p>
<hr>
<address>Apache Server at www.ros.org Port 80</address>
</body></html>

* Connection #0 to host www.ros.org left intact
```

```
$ curl -kv https://www.ros.org/reps/rep-0000.html
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 2605:bc80:3010:104::8cd3:962:443...
* Connected to www.ros.org (2605:bc80:3010:104::8cd3:962) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: C:/Program Files/Git/mingw64/ssl/certs/ca-bundle.crt
  CApath: none
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [93 bytes data]
* TLSv1.2 (IN), TLS handshake, Certificate (11):
{ [2558 bytes data]
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
{ [333 bytes data]
* TLSv1.2 (IN), TLS handshake, Server finished (14):
{ [4 bytes data]
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
} [70 bytes data]
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.2 (OUT), TLS handshake, Finished (20):
} [16 bytes data]
* TLSv1.2 (IN), TLS handshake, Finished (20):
{ [16 bytes data]
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: CN=www.ros.org
*  start date: Mar 24 05:33:31 2020 GMT
*  expire date: Jun 22 05:33:31 2020 GMT
*  issuer: C=US; O=Let's Encrypt; CN=Let's Encrypt Authority X3
*  SSL certificate verify ok.
} [5 bytes data]
> GET /reps/rep-0000.html HTTP/1.1
> Host: www.ros.org
> User-Agent: curl/7.69.1
> Accept: */*
>
{ [5 bytes data]
* Mark bundle as not supporting multiuse
< HTTP/1.1 403 Forbidden
< Date: Tue, 21 Apr 2020 04:03:52 GMT
< Server: Apache
< Content-Length: 282
< Content-Type: text/html; charset=iso-8859-1
<
{ [5 bytes data]
100   282  100   282    0     0    414      0 --:--:-- --:--:-- --:--:--   414<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>403 Forbidden</title>
</head><body>
<h1>Forbidden</h1>
<p>You don't have permission to access /reps/rep-0000.html
on this server.</p>
<hr>
<address>Apache Server at www.ros.org Port 443</address>
</body></html>

* Connection #0 to host www.ros.org left intact
```

knorth55, 2020-04-22:
I have same issue. I cannot access to https://www.ros.org/reps/rep-0003.html because of 403 access Forbidden.

Tiryoh, 2020-04-24:
Problem seems solved.
Thank you for your support.

closed 2020-04-24 by Tiryoh