ros-infrastructure/rep #304: xsd validation failing

tfoote, 2021-02-11:
Tonight the nightly build failed due to xsd validation failing. It doesn't look like a code change here but likely a library/dependency change, or a bug upstream that we need to pin until there's a fix.

https://build.ros.org/job/doc_independent-packages/630/console

```
02:10:46 rep-0124.rst (text/x-rst) -> rep-0124.html
02:10:47 rep-0119.rst (text/x-rst) -> rep-0119.html
02:10:47 rep-0000.rst (text/plain) -> rep-0000.html
02:10:47 python3 xsdValid.py
02:10:47 Traceback (most recent call last):
02:10:47   File "xsdValid.py", line 3, in <module>
02:10:47     import xmlschema
02:10:47   File "/usr/local/lib/python3.5/dist-packages/xmlschema/__init__.py", line 15, in <module>
02:10:47     from .resources import normalize_url, normalize_locations, fetch_resource, \
02:10:47   File "/usr/local/lib/python3.5/dist-packages/xmlschema/resources.py", line 13, in <module>
02:10:47     from elementpath import iter_select, Selector, XPath1Parser
02:10:47   File "/usr/local/lib/python3.5/dist-packages/elementpath/__init__.py", line 26, in <module>
02:10:47     from .xpath_token import XPathToken
02:10:47   File "/usr/local/lib/python3.5/dist-packages/elementpath/xpath_token.py", line 43, in <module>
02:10:47     from .tdop import Token, MultiLabel
02:10:47   File "/usr/local/lib/python3.5/dist-packages/elementpath/tdop.py", line 166
02:10:47     self._source: str = parser.source
02:10:47                 ^
02:10:47 SyntaxError: invalid syntax
02:10:47 Makefile:26: recipe for target 'xsdvalid' failed
02:10:47 make[1]: Leaving directory '/tmp/repositories/rep'
```

jacobperron, 2021-02-11:
Syntax for variable annotations was added in Python 3.6. Based on the backtrace, it looks like `elementpath` introduced a variable annotation and we are using Python 3.5.

jacobperron, 2021-02-11:
See https://github.com/ros-infrastructure/ros_buildfarm_config/pull/201 that proposes pinning `elementpath`.

closed 2022-02-16 by clalancette